rsasaki0109/lidar_localization_ros2 #5: IMU Orientation

Gaurav-Kapoor-InMach, 2021-09-17:
Hello Ryohei 
Do we need IMU data which has orientation in it or does your algorithm fuse the angular velocaity and linear acceleration from the raw IMU data? I am asking this because my IMU does not give out orientation data.

Gaurav-Kapoor-InMach, 2021-09-19:
Hello Again

I checked in your code that you require orientation from the IMU. But to which frame is it relative to? Or do you require absolute orientation w.r.t. ENU or NED frame?

Best regards
Gaurav

rsasaki0109, 2021-09-22:
The 9-axis imu composite in this package is based on LeGO-LOAM, so please check LeGO-LOAM for the sensor layout.
https://github.com/rsasaki0109/pcl_localization_ros2/blob/master/include/pcl_localization/lidar_undistortion.hpp#L17-L18

The implementation of the 6-axis imu-only composite is a bit difficult. I am not going to work on it now.

If your robot has wheel odometry, you can use some package to publish wheel odometry and dead reckoning by imu as /odom and input it to this package.

Gaurav-Kapoor-InMach, 2021-09-22:
Thanks for your reply

closed 2021-09-23 by Gaurav-Kapoor-InMach