rsasaki0109/lidar_localization_ros2 #9: Unable to compile on Galactic

MrOCW, 2022-07-24:
Hi, attempting `colcon build --symlink-install` returns:
```
pcl_localization_ros2/src/pcl_localization_component.cpp: In member function ‘virtual CallbackReturn PCLLocalization::on_activate(const rclcpp_lifecycle::State&)’:
/home/warthog_ws/warthog2/src/pcl_localization_ros2/src/pcl_localization_component.cpp:80:42: error: no matching function for call to ‘rclcpp_lifecycle::LifecyclePublisher<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >::publish(sensor_msgs::msg::PointCloud2_<std::allocator<void> >::SharedPtr&)’
   80 |     initial_map_pub_->publish(map_msg_ptr);
```

rsasaki0109, 2022-07-24:
This package is only tested and supported up to ros2 foxy. (I do hope to get ros2 humble support eventually.)
Perhaps ros2 galctic has had a major change in the rclcpp API.

Markurt, 2022-07-26:
I also met the same issue when trying to compile on Foxy. do you have any idea now? @MrOCW

MrOCW, 2022-07-26:
@Markurt I made some minor edits to load a pcd, it is working for me now.

rsasaki0109, 2022-07-26:
Sorry, I only tested this package with dashing...

Markurt, 2022-07-27:
> @Markurt I made some minor edits to load a pcd, it is working for me now.

I modified the pcl_localization_component.cpp:80 from `initial_map_pub_->publish(map_msg_ptr);` to `initial_map_pub_->publish(*map_msg_ptr);` and now I can successfully build, but when I try to launch, I cannot visualize the PCD map in Rviz until I launch again. May I know that if you meet the similiar problem before?

rsasaki0109, 2022-07-27:
Added support for foxy.
The problem with pcd not being visible was because param was not being read correctly and pcd was not being published.

PR
https://github.com/rsasaki0109/pcl_localization_ros2/pull/10

rsasaki0109, 2022-08-18:
The issue has been resolved, so close this issue.

closed 2022-08-18 by rsasaki0109