rsasaki0109/lidar_localization_ros2 #12: is adjustDistortion very important for localization?

oym1994, 2022-09-02:
Hi, thanks for your great job!

I have a question, is the adjustDistortion very important for an outdoor robot locolization? My robot currently has no imu and I wander what is the influence or improvement for this function.

Thanks for your attention and I am always looking forward to your response.

rsasaki0109, 2022-09-02:
Point cloud distortion correction is effective in cases of rapid rotation or when moving at high speed, such as in a car, and improves the accuracy of locazation estimation.
If locazation estimation does not work, it is a good idea to have an IMU on board.

Red
https://tech.tier4.jp/entry/2021/08/04/160000#%E5%8F%96%E5%BE%97%E3%81%97%E3%81%9F%E7%82%B9%E7%BE%A4%E3%81%8C%E6%AD%AA%E3%82%93%E3%81%A7%E3%81%84%E3%82%8B%E6%AD%AA%E3%81%BF%E8%A3%9C%E6%AD%A3

No comparative data is available.

oym1994, 2022-11-24:
Thanks for your kind response. If we have any experiment, we will upload it.