rsasaki0109/lidar_localization_ros2 #13: IMU Calibration

CtPeng-USTC-ND, 2022-09-08:
Thanks for your valuable work!

Currently, I am using the lidar undistortion function and embedding it into hdl_localization. However, when I turned IMU on and played my bag, the localization failed. I'm wondering if it's a problem with IMU calibration, have you done the calibration in the function (I didn't find the interface for setting the IMU parameters)?

rsasaki0109, 2022-09-08:
The lidar undistortion function in this package is based on LeGO-LOAM, so if you are not using the LiDAR with built-in IMU, you need to implement tf yourself.

https://github.com/RobustFieldAutonomyLab/LeGO-LOAM

CtPeng-USTC-ND, 2022-09-08:
> 

Many thanks!

rsasaki0109, 2022-09-10:
I would like to implement this conversion at some point, so I will reopen this issue once so I don't forget.

CtPeng-USTC-ND, 2022-09-13:
Got it. Looking forward to the new version, thanks.

rsasaki0109, 2023-01-01:
https://github.com/rsasaki0109/pcl_localization_ros2/pull/17

closed 2023-01-16 by rsasaki0109