rsasaki0109/lidar_localization_ros2 #16: Handling angular velocities

grassjelly, 2022-11-25:
Hi, thanks for releasing the great package. I've been tinkering with this and the results have been promising so far. However, I realized that the robot gets lost when turning on higher angular velocity. Just wondering if there's any parameter you can recommend to tweak to resolve this issue? Thank you!

rsasaki0109, 2022-11-25:
If you enter odometry (vehicle speed and imu) into lidarslam_ros2, odometry will be used as the initial value during scan matching, making it more resistant to rotation.
If Odometry is used, use_odom must be set to true.

For IMU only, there are currently no parameters for support.
If you want to improve it, you can update the following pose based on the IMU values.
https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/src/pcl_localization_component.cpp#L376

rsasaki0109, 2022-11-25:
Also, in general, GICP is more resistant to rotation than NDT.

rsasaki0109, 2023-07-19:
Since there are no activities, I will close this issue for now. If there is anything else, please reopen this issue.

closed 2023-07-19 by rsasaki0109