rsasaki0109/lidar_localization_ros2 #20: pose drift largely

pjs9115916, 2023-03-03:
[image]
I tested the localization algorithm in bag, I found the pose drift very large, I wonder whether my setup was not correct?  if I set the param of "set_initial_pose" to false, the path will disappear, so I have to set it to true.

pjs9115916, 2023-03-03:
In my test I did not use imu data
Here is my parms:

`/**:
    ros__parameters:
      registration_method: "GICP"
      ndt_resolution: 1.0
      ndt_step_size: 0.1
      transform_epsilon: 0.01
      voxel_leaf_size: 1.0
      scan_max_range: 100.0
      scan_min_range: 1.0
      scan_period: 0.1
      use_pcd_map: true
      map_path: "/home/terra/ros2_ws/src/pcl_localization_ros2/map/map.pcd"
      set_initial_pose: true
      initial_pose_x: 0.0
      initial_pose_y: 0.0
      initial_pose_z: 0.0
      initial_pose_qx: 0.0
      initial_pose_qy: 0.0
      initial_pose_qz: 0.0
      initial_pose_qw: 1.0
      use_odom: false
      use_imu: false
      enable_debug: false
`

rsasaki0109, 2023-03-03:
Indoors, the following parameters should be reduced.

ndt_resolution (0.5 - 2.0 [m] )
voxel_leaf_size(0.1~1.0[m])

Also, is the initial pose correct?

rsasaki0109, 2023-03-03:
By the way, which package and which lidar did you use to create the map point cloud?

pjs9115916, 2023-03-03:
> Indoors, the following parameters should be reduced.
> 
> ndt_resolution (0.5 - 2.0 [m] ) voxel_leaf_size(0.1~1.0[m])
> 
> It might also be a good idea to set registration_method to NDT.
> 
> Also, is the initial pose correct?

registration_method to NDT also will drift.

I record a bag use velodyne 16p. use your slam bag to create a pcd map,and then play the lidar bag to test the localization.

pjs9115916, 2023-03-03:
> By the way, which package and which lidar did you use to create the map point cloud?

I record a bag use velodyne 16p. use your slam bag to create a pcd map,and then play the lidar bag to test the localization.

rsasaki0109, 2023-03-03:
You might want to set voxel_leaf_size to 0.1 or 0.5.

rsasaki0109, 2023-03-03:
What is the SLAM you used, is it OSS?

You can share pcd and rosbag for me to see.
My email address is below.
rsasaki0109@gmail.com

pjs9115916, 2023-03-03:
> What is the SLAM you used, is it OSS?hdl_graph_slam?lio-sam?lidarslam_ros2?
> 
> You can share pcd and rosbag for me to see. My email address is below. [rsasaki0109@gmail.com](mailto:rsasaki0109@gmail.com)

lidarslam_ros2

rsasaki0109, 2023-03-03:
If lidarslam_ros2 is also parameterized for indoor use, a more accurate map can be produced.

pjs9115916, 2023-03-06:
> 

Hi author   have you received my pcd and rosbag?

rsasaki0109, 2023-03-06:
I have not received those yet.

pjs9115916, 2023-03-07:
> rsasaki0109@gmail.com

I sent you again few seconds ago

TheodorPatrickZ, 2023-03-27:
@rsasaki0109 were you able to reproduce the pose drift?

pjs9115916, 2023-04-10:
> The following parameters allow for some localization indoors. However, localization breaks down where there is not much of a point cloud If you add a time series filter such as Kalman filter, it may be less likely to fail.
> 
> ```miniyaml
> /**:
>     ros__parameters:
>       registration_method: "NDT"
>       ndt_resolution: 1.0
>       ndt_step_size: 0.1
>       transform_epsilon: 0.1
>       voxel_leaf_size: 0.1
>       scan_max_range: 10.0
>       scan_min_range: 0.1
>       scan_period: 0.1
>       use_pcd_map: true
>       map_path: "/home/autoware/Downloads/pcd_loc_ros2/map.pcd"
>       set_initial_pose: true
>       initial_pose_x: 0.0
>       initial_pose_y: 0.0
>       initial_pose_z: 0.0
>       initial_pose_qx: 0.0
>       initial_pose_qy: 0.0
>       initial_pose_qz: 0.0
>       initial_pose_qw: 1.0
>       use_odom: false
>       use_imu: false
>       enable_debug: true
> ```
> 
> I tried setting registration_method: "GICP" and reducing voxel_leaf_size in various ways, but it didn't seem to work.

thanks for your test and reply

rsasaki0109, 2023-12-02:
Using [ndt_omp](https://github.com/rsasaki0109/ndt_omp_ros2) has prevented loss of localization

[image]

closed 2023-12-02 by rsasaki0109