rsasaki0109/lidar_localization_ros2 #21: i can't view map

tazel7885, 2023-03-29:
https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION

i use map.pcd and rosbag file in above git link

when i launch your code i can't see map on rviz

what's your perception_pcl library? 
https://github.com/ros-perception/perception_pcl          <- is that??

rsasaki0109, 2023-03-29:
Is map_path set correctly? Please check it in the terminal output.

> map_path: "/home/sasaki/github_ws/ros2_ws/src/pcl_localization_ros2/map/bin_tc-2017-10-15--ndmap.pcd"
https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/param/localization.yaml#L12

>   RCLCPP_INFO(get_logger(),"map_path: %s", map_path_.c_str());
https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/src/pcl_localization_component.cpp#L178

rsasaki0109, 2023-03-29:
> what's your perception_pcl library?
https://github.com/ros-perception/perception_pcl <- is that??

Yes, this package is using pcl_conversions from perception_pcl.

tazel7885, 2023-03-30:
https://github.com/rsasaki0109/lidarslam_ros2

i make pcd map using above git. And data is hdl_400.bag

But my result is bad. 

i use parameter under!

is there any wrong?

/**:
    ros__parameters:
      registration_method: "NDT"
      ndt_resolution: 3.0
      ndt_step_size: 0.1
      transform_epsilon: 0.01
      voxel_leaf_size: 1.0
      scan_max_range: 100.0
      scan_min_range: 1.0
      scan_period: 0.1
      use_pcd_map: true
      map_path: "/home/tazel7885/ros2_ws/src/pcl_localization_ros2/map/map.pcd"
      set_initial_pose: true
      initial_pose_x: 0.0
      initial_pose_y: 0.0
      initial_pose_z: 0.0
      initial_pose_qx: 0.0
      initial_pose_qy: 0.0
      initial_pose_qz: 0.0
      initial_pose_qw: 1.0
      use_odom: false
      use_imu: true
      enable_debug: true

tazel7885, 2023-03-30:
[image]

tazel7885, 2023-03-30:
my result

tazel7885, 2023-03-30:
[image]

this is git result

rsasaki0109, 2023-03-30:
I'll check.

rsasaki0109, 2023-03-31:
`ndt_resolution` and `voxel_leaf_size` may be too large. Could you try with the following parameter set?

  | ndt_resolution | voxel_leaf_size
-- | -- | --
1 | 2.0 | 1.0
2 | 2.0 | 0.5
3 | 1.0 | 0.5
4 | 1.0 | 0.2

tazel7885, 2023-03-31:
i test all. but result is same.

Just in case, it seems that the map data and sensor data do not match inside the code. Is there any way to check if the map has been loaded from inside the code?

rsasaki0109, 2023-03-31:
> Is there any way to check if the map has been loaded from inside the code?

Currently, there is no such function, but you can have the message output at the following

https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/src/pcl_localization_component.cpp#L76-L96

rsasaki0109, 2023-03-31:
It was confirmed that localization is possible with the following parameters

```
/**:
    ros__parameters:
      registration_method: "GICP"
      ndt_resolution: 1.0
      ndt_step_size: 0.1
      transform_epsilon: 0.01
      voxel_leaf_size: 0.5
      scan_max_range: 100.0
      scan_min_range: 1.0
      scan_period: 0.1
      use_pcd_map: true
      map_path: "/home/map4/Downloads/hdl_data/map.pcd"
      set_initial_pose: true
      initial_pose_x: 0.0
      initial_pose_y: 0.0
      initial_pose_z: 0.0
      initial_pose_qx: 0.0
      initial_pose_qy: 0.0
      initial_pose_qz: 0.0
      initial_pose_qw: 1.0
      use_odom: false
      use_imu: false
      enable_debug: true
```

[image]

tazel7885, 2023-04-11:
Thanks!

I have some questions!

If I want to add odom information, can I add it in m units?

Have you ever turned the above algorithm into a camera point cloud?

rsasaki0109, 2023-04-11:
Odometry information needs to be input in meters according to the ROS specifications.
I have never used an input point cloud from a camera.

rsasaki0109, 2023-04-11:
I think you need to do a better job of parameter tuning because the accuracy of the scan point cloud is poor and the FoV is small with cameras compared to LiDAR.

rsasaki0109, 2023-05-04:
The issue regarding the title seems to have been resolved, so we will be closing this matter. If you have any further concerns, please feel free to open a new issue.

closed 2023-05-04 by rsasaki0109