rsasaki0109/lidar_localization_ros2 #22: Using odom

adiser, 2023-04-13:
Hi Ryohei, amazing work on the repo

I got pretty much everything to work robustly with just LIDAR. May I know what is the best way to integrate my own odometry source and the parameters to tune? I set the use_odom to True but it seemed like scan matching is not working

rsasaki0109, 2023-04-13:
It might be a bug, I'll take a look.

memo
pcl_localization_ros2 odom
https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/src/pcl_localization_component.cpp#L282-L323
lidarslam_ros2 odom
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/scanmatcher/src/scanmatcher_component.cpp#L319-L334

rsasaki0109, 2023-04-13:
@adiser 
By the way, can I ask you to share the data?

rsasaki0109, 2023-04-13:
I checked the data I have for bugs and there seemed to be no problem.
I have a feeling that there is a problem with the odometry in your system.
Does the trajectory with only odometry seem to have some problems?
Or can you share the data with me?

closed 2023-07-19 by rsasaki0109

robot-Yang, 2024-07-23:
Hi, @rsasaki0109, thanks for the nice project.
I successfully made the 3d localization with cloud-only work, and it seems nice, except a bit shaking of the path, so I tried to integrate odom and imu, but neither could improve the performance.
My odom data quality is not bad, at least it is stable although has a bit of drift after some rotation.
Do you have any idea? 
Thanks in advance!

rsasaki0109, 2024-07-23:
About `use_imu` optiom:
When 'use_imu' is set to true, did you also change the 'scan_period' (1/frequency)? If the results are still suboptimal, it's worth noting that the IMU fusion implemented in this package is based on the scan distortion correction from lego-loam. However, it doesn't utilize individual timestamps for each point, which can degrade performance. Thus, it might be better to set 'use_imu' to false.

About `is_odom` option:
Is the odometry input you are using fused with the IMU data? If it's derived from wheel-based angular velocity, the performance could be poor and might even degrade the scan matching.

robot-Yang, 2024-07-23:
Thanks for the quick reply.

Ok, I understand, it's better to set use_imu as false.

Yes, I am using the wheel odometry, I don't understand why it would degrade the performance, anyway, how about using other odometry source instead of the wheel-based one such as Fast-lio2, is it possible to enhance the performance with that?

rsasaki0109, 2024-07-23:
No, the wheel speed vx from wheel odometry is good, so you should fuse the wheel speed vx and the angular velocity from the IMU to create the odometry.

Refer to the following for guidance.
https://github.com/autowarefoundation/autoware.universe/tree/main/localization/gyro_odometer