rsasaki0109/lidar_localization_ros2 #23: cannot see the map in rviz

naiyisiji, 2023-04-18:
I add my own pcd map to the map_path parameter but I cannot see my map in rviz. I also play my dataset.db3 but I cannot see the path.  
Could you tell me what happened? Thank U so much!

rsasaki0109, 2023-04-18:
@naiyisiji 
Does the same problem occur if I launch this package after launching rviz?

naiyisiji, 2023-04-18:
Sorry I cannot open your dataset link, I only run my dataset and pcd map.

rsasaki0109, 2023-04-18:
@naiyisiji 
Can you confirm that the `initial_map` topic is published when this package is started? ros2 topic echo initial_map --no-arr.

If it is not published, the value of map_path in the yaml may be incorrect.

naiyisiji, 2023-04-18:
Hold on. I will test it soon.

naiyisiji, 2023-04-18:
Now my map is shown in rviz. However I face a new problem. When I use 
ros2 launch pcl_localization_ros2 pcl_localization.launch.py 
I cannot replay my dataset xxxx.db3, which means there are no more message show in terminal ros2 topic hz mytopic. I think my topic message may be blocked. Do you know what happened?

rsasaki0109, 2023-04-18:
I think rosbag is broken.

naiyisiji, 2023-04-18:
I am sure my rosbag is nor broken QAQ
x-special/nautilus-clipboard
copy
file:///home/nasky/%E6%A1%8C%E9%9D%A2/254_1681811164_hd.jpeg

naiyisiji, 2023-04-18:
[image]

naiyisiji, 2023-04-18:
Thank you Sir. The problem is solved.

closed 2023-04-18 by naiyisiji

GauleeJX, 2024-11-06:
> @naiyisiji If you run this package after you run rviz, will you have the same problem?

My rviz crashed after running this package. Error like this:rviz2: /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.8/.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/include/OgreAxisAlignedBox.h:225：void Ogre::AxisAlignedBox::setExtents(const Vector3&, const Vector3&): 假设 ‘(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"’ 失败。

And map is loaded successfully, because terminal returns:
lidar_localization]: Map Size 395053
[lidar_localization_node-1] [INFO] [1730910708.256110969] [lidar_localization]: Initial Map Published
[lidar_localization_node-1] [INFO] [1730910708.269485218] [lidar_localization]: Activating end
But ros2 topic echo /initial_map return nothing, wh

rsasaki0109, 2024-11-07:
If rviz2 crashes, the configuration file might be corrupted. Try deleting the configuration file and setting up the visualization manually. 
It may also help to reinstall rviz2 with the --reinstall option using apt install.