rsasaki0109/lidar_localization_ros2 #24: The agent cannot turn left or right durning rosbag replay

naiyisiji, 2023-04-21:
Hi there. When I ros2 bag play my_dataset.db3, I found the pose cannot turn left or right and draft largely. Do you know why the drift happen? The video recording the draft shows the problem, you can see the red pose vector never change.

[20-04-23 屏幕视频 18:05:55.webm](https://user-images.githubusercontent.com/89254461/233551233-fbce0f25-a3bb-432f-9906-729e31403cf6.webm)

rsasaki0109, 2023-04-22:
What are the parameters like?
The following parameters might be good for outdoor use.

| registration_method | ndt_resolution | voxel_leaf_size |
| --- | --- | --- |
| NDT| 2.0 | 1.0 |
| GICP| - | 2.0 |
| GICP| - | 1.0 |
| GICP| - | 0.5 |

Can you visualize the map point cloud as well?

naiyisiji, 2023-04-22:
> This email originated from outside of City University of Hong Kong. Do not reply, click links, or open attachments unless you confirm the identity of the sender and know that the content is safe.

Dear Sasaki,

Thank you so much for you help! Now I have already solve this problem by changing the initial parameters.
Your program is really useful!

Regards,
Zhao Guanyi

closed 2023-04-23 by rsasaki0109