rsasaki0109/lidar_localization_ros2 #25: specifying the lidar frame id

space192, 2023-06-20:
Is it possible to set the frame id of the lidar in the parameters.yaml file ?

rsasaki0109, 2023-06-20:
The following frame IDs can be set, but there is no transformation from the scan's lidar frame to the base_link.
https://github.com/rsasaki0109/pcl_localization_ros2/blob/main/src/pcl_localization_component.cpp#L9-L11

If needed, you might have to implement it referring to the following:
https://github.com/autowarefoundation/autoware.universe/blob/main/localization/ndt_scan_matcher/src/ndt_scan_matcher_core.cpp#L492-L506C1

rsasaki0109, 2026-10-10:
Closing: the LiDAR frame no longer has to match `base_link`. The localizer takes the LiDAR frame from the cloud's `header.frame_id` and transforms the cloud into `base_frame_id` through TF (`src/component_scan_preparation.cpp`). `quickstart.py` accepts `--lidar-frame` and `--[no-]publish-lidar-tf`, and warns when the `base -> LiDAR` transform is missing; see [docs/quickstart.md](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/quickstart.md#other-lidar-drivers).

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-10 by rsasaki0109