rsasaki0109/lidar_localization_ros2 #26: Localization worser then in your lidarslam_ros2

SamuelHafner, 2023-07-25:
Hi @rsasaki0109, 

thank you for your great work. I have a question.

We used your lidarslam_ros2 for mapping and localization and used then your pcl_localization_ros to test the localtization in the map. With the same rosbag and the map from lidarslam_ros2 as input. 
Why is the localization so much worser. Did we forget something? 

Thank you in advanced

Greeting
Samuel

rsasaki0109, 2023-07-26:
To achieve good localization, adjustments such as ndt resolution, voxel grid filter parameters, etc. are necessary. Please refer to previous issues.

SamuelHafner, 2023-07-26:
Okay perfect thank you. I have tried some other parameters, its a little bit better. 
I will publish the results if I found a good parameters for indoor and outdoor localization.

rsasaki0109, 2023-12-02:
I recently implemented it in a PR, but using NDT_OMP/GICP_OMP instead of NDT/GICP might improve accuracy. 

As there seems to be no activity, I will temporarily close this issue. Please open a new issue if there's anything further.

PR
https://github.com/rsasaki0109/pcl_localization_ros2/pull/31

closed 2023-12-02 by rsasaki0109