rsasaki0109/lidar_localization_ros2 #33: about ros2 humble

lidarmansiwon, 2024-03-05:
hi rsasaki, I'm getting a lot of help from you
while attempting to use, The part of launch file, remapping ouster/points to /cloud. because I want to use ouster. but It didn't work. it can't get point cloud. so I found code that subcribe pointcloud in pcl_localization_component.cpp. In that code, I can find "velodyne_points". so I edit launch file. change /cloud to /velodyne_points. after edit. It work. but it can't see work well. It's very inaccurate. please help me. 
[image]
[image]
[image]

rsasaki0109, 2024-03-07:
Please tell me if the input cloud topic is set to /ouster/points when running ros2 node info /lidar_localization. 
Also, please describe how the terminal output looks for lidar_localization_ros2.

rsasaki0109, 2024-03-07:
Looking at your comment again, it seems that the scan matching itself is being performed, isn't it? What do you mean by 'inaccurate'? I need a bit more detailed information. If the scan matching is failing, it may be necessary to adjust the ndt_resolution and voxel_leaf_size.

youweik2, 2024-11-01:
Have you solved this issue? I’d appreciate it if you could share your solution.

rsasaki0109, 2026-10-10:
Closing this old question. Humble is supported (Jazzy is the primary target). An Ouster topic can be selected directly without editing the source, for example `quickstart.py --map /path/to/map.pcd --cloud-topic /ouster/points` ([docs/quickstart.md](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/quickstart.md#other-lidar-drivers)). For inaccurate matching, [docs/troubleshooting.md](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/troubleshooting.md) has a checklist covering map alignment, the `base -> LiDAR` transform, and `/alignment_status`. Please open a new issue with that output if it still fails on a current build. The Ouster-specific discussion continues in #34.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-10 by rsasaki0109