rsasaki0109/lidar_localization_ros2 #34: Use diffirent sensor like Ouster or etc

Mustafa19077, 2024-03-21:
Hi rsasaki, I have a problem. Localization is fine when ı used hdl_400 bag file. But there is a problem when ı use ouster lidar and ouster bag file. I have same issue with lidarslam_ros2 but ı have solved this issue by changing "base_link" to "os_lidar" (inside params) , and "velodyne" to "os_lidar", "velodyne_points" to "ouster/points". 
 with lidar_localizaiton, ı make the same change but didnt work this time. This is the general information about my problem.
I want to ask what kind of change that ı should when ı use different sensor like ouster.

rsasaki0109, 2024-03-22:
I think it will work if you change the frame id from `velodyne` to the one for ouster/points. You can see the frame id by running 「ros2 topic echo ouster/points --no-arr」.

https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/launch/lidar_localization.launch.py#L27

youweik2, 2024-11-01:
Have you solved this issue? I’d appreciate it if you could share your solution.

rsasaki0109, 2024-11-01:
I feel like I spoke with him via email or DM, but I haven't heard any solutions from him yet.

Mustafa19077, 2024-11-01:
Hello, I solved this problem. Here is how ı did this.
Lidars topic is ouster/points. But in localization settings, it is velodyne/points or like that. When ı changed the as ouster/points it didnt resolved it. So, I write a python function that converts the topic. You write a python code that subcsribes that ouster/points and publish as velodyne/points. After this step you tje locallization will work. I tried both slam and localitioz library in my real car, it perfectly works. I hope this solution helps you.

İf you have any problem about slam or localization please let me know.

youweik2, 2024-11-05:
> Hello, I solved this problem. Here is how ı did this. Lidars topic is ouster/points. But in localization settings, it is velodyne/points or like that. When ı changed the as ouster/points it didnt [...]

It works, thank you so much!

csd-sudo, 2025-05-15:
@youweik2 Why not directly modify the name of the topic that accepts radar data in the code? Will this affect the performance?

rsasaki0109, 2026-10-09:
On current `main` / v1.3.0, you can select the driver's `sensor_msgs/msg/PointCloud2` topic directly; no relay is needed just to rename it:

```bash
ros2 run lidar_localization_ros2 quickstart.py --map /path/to/map.pcd --cloud-topic /your_lidar/points
```

Use the cloud's `header.frame_id` and provide the calibrated TF to the robot base. See [the driver setup notes](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/quickstart.md#other-lidar-drivers); Hesai hardware has not been validated here.

closed 2026-10-27 by rsasaki0109