rsasaki0109/lidar_localization_ros2 #35: the direct positioning got stuck

ming1j2, 2024-04-09:
Hello, I would like to ask why I entered a source point cloud and opened the real-time point cloud of the radar. However, during operation, there was a lag phenomenon during real-time radar positioning, and in the end, the direct positioning got stuck

rsasaki0109, 2024-04-15:
Sorry, I just noticed your comment.
I would like to know more details. For example, whether the environment is indoors or outdoors, or if you have done any parameter tuning for that.
Also, I would like the YAML file. It would be better if there are images or videos of the position estimation.
Setting ndt_num_threads to 0 (meaning maximum allowable threads are used) can sometimes cause problems, so it might be good to adjust it.

ming1j2, 2024-04-17:
In the indoor environment, my yaml file only changed the input path of the source point cloud for map_path and the number of ndtnum_threads to 4. Other changes may have been made to the point cloud topic and imu topic. The effect is that after opening it, it can be repositioned and registered, but after a period of time, it will occasionally get stuck, and in the end, the positioning will just get stuck. I don't know if there was a problem with the change

ming1j2, 2024-04-17:
And there is also a strange issue, which is that after changing the number of ndtnum_threads to 4, but the info display result of ROS2 is still 0, which makes me not sure if the subsequent repositioning has undergone multi-threaded acceleration

rsasaki0109, 2024-04-17:
In ROS2, you need to rebuild to apply parameter changes.
If you add the --symlink-install option during the build, you no longer need to rebuild.

ming1j2, 2024-04-17:
But what is the problem with this repositioning getting stuck in the end? Did you encounter this issue while trying out the indoor scene radar

rsasaki0109, 2024-04-17:
I'm not quite sure what the problem is, so I'd like to see an image or video of rviz.

ming1j2, 2024-04-17:
[image]

ming1j2, 2024-04-17:
[image]

ming1j2, 2024-04-17:
[image]

ming1j2, 2024-04-17:
[image]

ming1j2, 2024-04-17:
Just like this

rsasaki0109, 2024-04-17:
Can you upload the rviz screen? I'd like to see the map, trajectory, scans, etc.

rsasaki0109, 2024-04-17:
The delta angle is large.
By the way, have you been swinging the LiDAR around by hand?
The current LiDAR localization uses IMU coupling based on LeGO-LOAM which might be quite sensitive to that.

ming1j2, 2024-04-17:
Yes, I did swing the LiDAR with my hand. Can't I swing it with my hand? If rviz is turned on, then the trajectory is the one I swing the radar with my hand