rsasaki0109/lidar_localization_ros2 #37: CPU占用率爆炸

ChangGL, 2024-05-07:


rsasaki0109, 2024-05-07:
The cause might be that ndt_num_threads is set to 0. Please adjust it according to your PC specs.
https://github.com/rsasaki0109/lidar_localization_ros2/tree/main?tab=readme-ov-file#params

ChangGL, 2024-05-09:
Can this package be adapted to the ndt_gpu method to reduce CPU usage?

rsasaki0109, 2024-05-09:
I think it can be.

rsasaki0109, 2026-10-27:
On current `main`:

- `ndt_num_threads` now defaults to `4` in the node and in `param/localization.yaml`. `0` means all OpenMP threads, which was the cause here. `create_lidar_localization_config.py --ndt-threads N` sets it for generated configs; [mid360_legged_jetson.md](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/mid360_legged_jetson.md) uses `2` on a Jetson.
- Further CPU levers are `voxel_leaf_size`, `scan_max_range`, and, on large maps, the local map crop with `local_map_update_distance`, so the target is not rebuilt every scan. See [Pose Rate Drops on Dense Maps](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/troubleshooting.md#symptom-pose-rate-drops-or-tracking-is-lost-on-dense-maps).
- The quickstart status line reports `Alignment is falling behind` (cause `overload`) when matching gets slower than the scan rate.

A GPU NDT backend is not planned for now. Closing; please reopen with your thread count and `/alignment_status` if CPU use is still too high on current `main`.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-27 by rsasaki0109