rsasaki0109/lidar_localization_ros2 #39: Localization does not work after some time

shavargo109, 2024-06-19:
At first it works smoothly, but at some time when I received the following warning,
`Message Filter dropping message: frame 'velodyne' at time 1712028980.643 for reason the queue is full`

the program just stuck at there and did not publish any pcl_pose at all, i wonder each part of the code cause the problem

I tried with 2 different computers, the computer with less computing power works fine but not well for that of a computer with a better computing power

rsasaki0109, 2024-06-20:
Here is the link that might be helpful:

[https://get-help.theconstruct.ai/t/rviz-node-message-filter-dropping-message-frame-base-scan-at-time-701-469-for-reason-discarding-message-because-the-queue-is-full/29097/4](https://get-help.theconstruct.ai/t/rviz-node-message-filter-dropping-message-frame-base-scan-at-time-701-469-for-reason-discarding-message-because-the-queue-is-full/29097/4)

shavargo109, 2024-06-24:
Thanks! It turns out to be a ros2 bag issue, when i tried with actual sensor data, it works perfectly fine!

closed 2024-06-24 by shavargo109