rsasaki0109/lidar_localization_ros2 #42: map to odom tf instead of map to base link tf

arwinder-jaspal, 2024-07-15:
Is it possible to get the map to odom tf instead ? this is so that i can use this with Nav2. Or if anyone has any experience making nav2 work as it is please let me know

Hailemichael12m, 2024-07-15:
Hi, I have been using this localization with Nav2. Nav2 doesn't specifically require map to odom transform. If you just stop broadcasting odom to base_link transform and instead use the map to base_link transform published by lidar_localization_ros2, you can use it with Nav2. Your tf tree will then look like map -->base_link instead of map-->odom-->base_link.

arwinder-jaspal, 2024-07-15:
Does this mean for your local costmap you set your global frame to map instead of odom ? This is however violating REP 105

karthist, 2024-08-08:
Hi, I am working on a similar goal. Were you able to use a map to odom transform instead of the map to base_link transform? Did it affect your performance?

Hailemichael12m, 2024-08-12:
Hi, I also tried to use map to odom transform (this also needed setting use_odom to false) but I didn't see a significant difference in performance. I still see some drift in both cases.

karthist, 2024-08-12:
Hi, thanks for responding. Can you explain how you achieved this in a little more detail? What changes did you have to do the original implementation? Additionally, you mentioned you used it with nav2. How did you integrate this setup with the robot_localization package and the rest of nav2? I appreciate your insights and help.

karthist, 2024-08-15:
@rsasaki0109 Hi, do you have any inputs on this? I am trying to integrate this setup with nav2. Do you have any recommendations on how I can use this to publish a map to odom and transform, and combine it with the robot_localization package to get odom to base_link transform?

arwinder-jaspal, 2024-08-16:
you dont need map to odom tf for nav2, there will be some tuning on the nav2 side but yes it does work

closed 2024-08-16 by arwinder-jaspal

karthist, 2024-08-16:
I understand that, but wouldn't it violate REP-105 if we do that? Also, what kind of tuning are you talking about?

Hailemichael12m, 2024-08-18:
Sorry for my late response. Since nav2 takes 2D map only, you need to align the 3D .pcd map with the 2D map you use in navigation. One way to do that is to generate a 2D map from the same 3D map you use in localization. Then from nav2 side, use map as a fixed frame in local costmap.

kzm784, 2024-09-09:
I am using nav2 and lidar_localization_ros2 for navigation with the same procedure, but the local_costmap is not displayed at a certain height, probably because of the self-location estimation in the 3d map.I would like to know if you have the same symptom or solution.