rsasaki0109/lidar_localization_ros2 #47: error, rviz no map frame. and if give 2D pose, the node die.

Un1Lee, 2024-11-07:
I really need your help....
I use fastlio2 for mapping, then get pcd map.
the fastlio publish tf: odom->base_link.
when I run your code, I use odom: false, use imu: false.
I run mid360, and fastlio2 will transform livox -> pointcloud2 form. it has 2 topic, one for frame_id odom, and one for frame_id base_link. I check both for your code,  both are not work. **In rviz, it show no map frame in global stats: fixed frame**.
I have deleted your remap, if I run rqt_tf_tree, it will see odom->base_ink->velody(I have change it to lidar_link). but no map. I am confused that your code would publish map->odom, but it dose not be saw.
When I see rviz, it can see my map. but no lidar cloud. (what's more, this lidar cloud can be saw in new rviz).  And if I give it initial pose, the it will report posereceived, cloudreceived, then node die. I use info to know maybe it is because pcl::fromROSmsg, but it really no reason to die for this. 

Thank you for your help.

rsasaki0109, 2024-11-11:
Perhaps the map_path setting in yaml isn't correct?
Can you check if the PCD map is correct by running head <MAP_PATH>?
If that looks fine, it might help if you could share the terminal output when you run it.

Mutant0720, 2024-12-08:
When I do that with mid360 I found the same error and I solved it by remapping the topic name.
pcl_localization node need to subscribe velodynepoints but for mid360 you should use /livox/lidar.
you can check the topic names by
$ros2 node info /pcl_localization

rsasaki0109, 2026-06-10:
Fixed on main (4cd15f0): /initialpose now publishes map TF immediately; frame mismatch and map-not-ready warnings are explicit. See docs/frame_contract.md RViz checklist and docs/troubleshooting.md.