rsasaki0109/lidar_localization_ros2 #48: map showing up on rviz alongside live data

Matttran7, 2024-11-12:
Hello,
I have gotten the package to work with the Livox Avia using Ros2 foxy (seeing the vector transformations in terminal), however I was wondering if there was a way to overlay the initial map on top of it to help visualize the localization process. In addition, do you have any recommendations on the YAML configurations for slow and indoors?

Thanks

rsasaki0109, 2024-11-12:
You should be able to visualize it in RViz. Let’s check the map topic displayed in RViz.

Indoors, an ndt_resolution of 0.5 to 1 and a voxel_leaf_size of around 0.1 to 0.2 should work well. There aren't many parameters that need to be adjusted for low-speed operations. This package was designed for 360-degree rotating LiDAR, so it might be challenging to use it for self-localization with the Livox Avia.

rsasaki0109, 2024-11-18:
Ah, with the current settings, the map might only be published at the time the node is launched. How about launching the node after starting RViz?

rsasaki0109, 2025-05-15:
I’m not sure I understand your question. Did you mean to ask, “Is this package not suitable for the Livox MID360 LiDAR?” I haven’t tried it myself, but I’ve seen people using it.
Here’s one example: https://www.youtube.com/watch?v=aOVUDg7gij8