rsasaki0109/lidar_localization_ros2 #50: Enhance the stability of the program

YIYuNCU, 2024-12-03:
The stability of the current code is not very well. If there is an error exception, it will directly lead to the termination of the node. Is it possible to add some "try and catch" to catch the exception, to avoid the program from terminating because of the exception to make the program better applicable to the actual environment? (I'm sorry for my poor English if there are any grammatical or sentence sentiment issues)

rsasaki0109, 2024-12-03:
Thank you for the good issue! You’re absolutely right that there’s room for improvement, but unfortunately, I’m currently tied up with other tasks and may not be able to address this right away. Please feel free to share any further suggestions or insights in the meantime.

rsasaki0109, 2026-10-27:
Current `main` handles the failure paths that used to take the node down, instead of terminating:

- exceptions from `align()` are caught and the scan is rejected;
- TF lookups (scan, IMU, odom) are caught and logged;
- non-finite twist, IMU, odometry, registration fitness and poses are rejected, keeping the last valid pose;
- a map that fails to load or has no points (including `/map_switch`) leaves the current map;
- lifecycle activation goes through bounded `GetState`/`ChangeState` calls and fails visibly.

No specific crash location was reported here, so I'm closing this. If the node still terminates on current `main`, please open a new issue with the log and the last messages before the exit.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-27 by rsasaki0109