rsasaki0109/lidar_localization_ros2 #52: different results

oliaiaia, 2025-03-11:
Hi!
I recorded a bug file on the robot and play it back without changing the localization parameters. I stop the node each time, then start it. The first launch produced an adequate result, the route matched the expected one, but then with each subsequent launch the result becomes worse and worse: see photo (haven't done photo of the 1st try). After rebooting the laptop, everything became really bad... 

2:
[image]
3:
[image]
4:
[image]
5:
[image]

`
/**:
    ros__parameters:
      registration_method: "NDT_OMP" 
      score_threshold: 0.05
      ndt_resolution: 3.0
      ndt_step_size: 0.1
      ndt_num_threads: 4
      ndt_max_iterations: 200
      transform_epsilon: 0.01
      voxel_leaf_size: 0.15
      scan_max_range: 100.0
      scan_min_range: 0.2
      scan_period: 0.1
      use_pcd_map: true
      set_initial_pose: true
      initial_pose_x: 266.99551
      initial_pose_y: -112.886769087
      initial_pose_z: -0.046483935
      initial_pose_qx: 0.00141948487108
      initial_pose_qy: -0.0049822938747
      initial_pose_qz: 0.751074512341
      initial_pose_qw: 0.66019712111
      base_frame_pose_x: 1.253
      base_frame_pose_y: 0.0
      base_frame_pose_z: 2.245
      base_frame_pose_qx: 0.0
      base_frame_pose_qy: 0.0
      base_frame_pose_qz: 0.0
      base_frame_pose_qw: 1.0

      use_odom: false
      use_imu: false
      enable_debug: true
      global_frame_id: map
      odom_frame_id: odom
      base_frame_id: rslidar #was base_link 

`

using only 3d lidar (rslidar), no imu, no odom. why may it be so?

rsasaki0109, 2025-03-11:
Is the CPU load okay? You can check it with the htop command. Other processes might be causing issues.

oliaiaia, 2025-03-12:
yes, the CPU is normal, I checked it at every launch 
Here is an example of the result during one of the unsuccessful launches:
[image]

rsasaki0109, 2025-03-12:
I'm curious about the terminal output when a failure occurs. It should be checked whether the issue lies in the QoS or DDS topic communication or if the scan matching processing time is slow.

For DDS, you may find the following reference helpful:
https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/

oliaiaia, 2025-03-12:
I use standard dds, tried rmw_cyclonedds_cpp, but the data from the point cloud did not arrive, probably because the robot had a bug file recorded with the standard

[output.txt](https://github.com/user-attachments/files/19203254/output.txt)

TPODAvia, 2025-07-14:
@oliaiaia the package expected to subscribe the BEST_EFFORT topic. If your lidar are publishing in RELIABLE qos - Modifyting the rclcpp::SensorDataQoS() to the rclcpp::SystemDefaultsQoS() will sove the issue

```bash
yhs@yhs-ros2:~/Orca_Robot/colcon_ws$ ros2 topic info /lidar/points2_raw -v
Type: sensor_msgs/msg/PointCloud2

Publisher count: 1

Node name: rslidar_points_destination_0
Node namespace: /rslidar_sdk
Topic type: sensor_msgs/msg/PointCloud2
Endpoint type: PUBLISHER
GID: 01.10.dc.3f.b5.57.1e.97.e2.da.6a.36.00.00.15.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (100)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: ndt_localization
Node namespace: /
Topic type: sensor_msgs/msg/PointCloud2
Endpoint type: SUBSCRIPTION
GID: 01.10.ef.80.12.26.3a.c0.86.9f.07.08.00.00.2b.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (5)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

```

rsasaki0109, 2026-10-27:
Two things on current `main` help narrow down run-to-run differences on a replayed bag:

- Start `ros2 bag play --clock` before quickstart, so localization runs on the bag's time (quickstart prints `clock=sim`).
- The status line reports when tracking weakens and why: few scan points, poor match, or CPU overload. See [reading the status line](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/troubleshooting.md#reading-the-status-line).

On QoS: the cloud subscriber is best-effort, which also receives from a reliable publisher, so no change is needed there.

Closing as stale. If runs still diverge on current `main`, please reopen with the quickstart output and `/alignment_status` from a bad run.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-27 by rsasaki0109