rsasaki0109/lidar_localization_ros2 #53: Why the performance of the localization with IMU is bad?

InguChoi, 2025-03-27:
Thank you for sharing your lidar_localization_ros2 package!

I test your package with our custom robot & ros2 bag file. Your localization method is based on map matching with IMU distortion.
So, I have a point cloud map made by SLAM. Then I put the point cloud map & execute the package.
I have two issues.

1.  It is very vulnerable to rotational motion estimation without IMU sensor. The rotational speed is not fast, but it cannot estimate the position even in slow-rotating motion.
[image]

2. So, I added IMU sensor information to make it robust to rotational motion. However, the estimation performance was significantly lower. It doesn't seem to be able to move from the starting point in the first place.
[image]

rsasaki0109, 2025-03-29:
The IMU fusion in this package is based on LEGO LOAM, but the method is quite outdated and not very effective. LIO-SAM is easy to implement and performs well, but we haven't integrated it yet. Referring to the issues in LEGO LOAM might be helpful

InguChoi, 2025-03-31:
@rsasaki0109 Thank you!

closed 2025-03-31 by InguChoi