rsasaki0109/lidar_localization_ros2 #54: corrent_pose_with_cov_stamped_ptr_未加锁

ccqedq, 2025-05-12:
odomReceived和cloudReceived两个回调函数同时使用了corrent_pose_with_cov_stamped_ptr_，为什么代码里不加锁呢

rsasaki0109, 2025-05-12:
(Since Autoware's localization module doesn't really use mutexes, maybe it's not necessary.)

rsasaki0109, 2026-10-10:
Closing: the race described here cannot happen in the current code. The node runs on a `MultiThreadedExecutor`, but `odomReceived` and `cloudReceived` both stay in the default mutually exclusive callback group, so they never run in parallel. The callbacks that do have their own groups take `callback_state_coordinator_`'s state lock before touching the shared pose state; see the comment on `corrent_pose_with_cov_stamped_ptr_` in `include/lidar_localization/lidar_localization_component.hpp`. Those callbacks are `/initialpose` and the opt-in pose-publication timer. Please reopen with a reproduction if you see a concurrent access.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-10 by rsasaki0109