rsasaki0109/lidar_localization_ros2 #55: odom_frame_id_    is defined but not used

little-lucky-bai, 2025-05-22:
odom_frame_id_ is defined but not used, and there is no published transformation between map and odom.

rsasaki0109, 2025-05-26:
Thanks for the report.
odom_frame_id_ is currently unused, and I don’t have plans to handle the map–odom transform at this time due to limited time.

rsasaki0109, 2026-10-10:
Closing as resolved: `odom_frame_id_` is used now. With `enable_map_odom_tf: true` the localizer looks up `odom_frame_id -> base_frame_id` and publishes `map -> odom` (`src/component_pose_publisher.cpp`). The two TF modes are described in [docs/frame_contract.md](https://github.com/rsasaki0109/lidar_localization_ros2/blob/main/docs/frame_contract.md).

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-10 by rsasaki0109