rsasaki0109/lidar_localization_ros2 #56: lidar_localization_ros2 crashed in the case of use_odom:=true

atinfinity, 2025-05-23:
## Summary

`lidar_localization_ros2` crashed in the case of `use_odom:=true`.

## Environment

- Ubuntu 24.04, x86_64
- ROS 2 Jazzy
- gcc13.3.0 

## Steps to reproduce

```bash
mkdir -p localization_ws/src
cd localization_ws/src
git clone https://github.com/rsasaki0109/ndt_omp_ros2.git
git clone https://github.com/rsasaki0109/lidar_localization_ros2.git
cd ..
rosdep install -y -i --from-paths src
colcon build --symlink-install
source install/setup.bash
```

I changed `use_odom` to `true`.

```bash
ros2 launch lidar_localization_ros2 lidar_localization.launch.py
```

The following message was displayed.

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lidar_localization_node-1]: process started with pid [53902]
[lidar_localization_node-1] [INFO] [1747976542.620000612] [lidar_localization]: Configuring
[lidar_localization_node-1] [INFO] [1747976542.620060586] [lidar_localization]: initializeParameters
[lidar_localization_node-1] [INFO] [1747976542.620080913] [lidar_localization]: global_frame_id: map
[lidar_localization_node-1] [INFO] [1747976542.620082979] [lidar_localization]: odom_frame_id: odom
[lidar_localization_node-1] [INFO] [1747976542.620084870] [lidar_localization]: base_frame_id: base_link
[lidar_localization_node-1] [INFO] [1747976542.620086587] [lidar_localization]: registration_method: NDT_OMP
[lidar_localization_node-1] [INFO] [1747976542.620088274] [lidar_localization]: ndt_resolution: 1.000000
[lidar_localization_node-1] [INFO] [1747976542.620091892] [lidar_localization]: ndt_step_size: 0.100000
[lidar_localization_node-1] [INFO] [1747976542.620094336] [lidar_localization]: ndt_num_threads: 4
[lidar_localization_node-1] [INFO] [1747976542.620096111] [lidar_localization]: transform_epsilon: 0.010000
[lidar_localization_node-1] [INFO] [1747976542.620098124] [lidar_localization]: voxel_leaf_size: 0.200000
[lidar_localization_node-1] [INFO] [1747976542.620099974] [lidar_localization]: scan_max_range: 40.000000
[lidar_localization_node-1] [INFO] [1747976542.620102448] [lidar_localization]: scan_min_range: 1.000000
[lidar_localization_node-1] [INFO] [1747976542.620104271] [lidar_localization]: scan_period: 0.100000
[lidar_localization_node-1] [INFO] [1747976542.620106187] [lidar_localization]: use_pcd_map: 1
[lidar_localization_node-1] [INFO] [1747976542.620114034] [lidar_localization]: map_path: /home/dandelion/map.pcd
[lidar_localization_node-1] [INFO] [1747976542.620115819] [lidar_localization]: set_initial_pose: 1
[lidar_localization_node-1] [INFO] [1747976542.620117408] [lidar_localization]: use_odom: 1
[lidar_localization_node-1] [INFO] [1747976542.620119086] [lidar_localization]: use_imu: 0
[lidar_localization_node-1] [INFO] [1747976542.620120655] [lidar_localization]: enable_debug: 1
[lidar_localization_node-1] [INFO] [1747976542.620123130] [lidar_localization]: initializePubSub
[lidar_localization_node-1] [INFO] [1747976542.624499039] [lidar_localization]: initializePubSub end
[lidar_localization_node-1] [INFO] [1747976542.624530115] [lidar_localization]: initializeRegistration
[lidar_localization_node-1] [INFO] [1747976542.624571155] [lidar_localization]: initializeRegistration end
[lidar_localization_node-1] [INFO] [1747976542.624575132] [lidar_localization]: Configuring end
[INFO] [launch.user]: -- Inactive --
[lidar_localization_node-1] [INFO] [1747976542.626323563] [lidar_localization]: Activating
[lidar_localization_node-1] [INFO] [1747976542.626343201] [lidar_localization]: initialPoseReceived
[lidar_localization_node-1] [INFO] [1747976542.626391640] [lidar_localization]: initialPoseReceived end
[lidar_localization_node-1] Failed to find match for field 'intensity'.
[lidar_localization_node-1] [INFO] [1747976542.634082133] [lidar_localization]: Map Size 362013
[lidar_localization_node-1] [INFO] [1747976542.641433950] [lidar_localization]: Initial Map Published
[lidar_localization_node-1] [INFO] [1747976542.652082203] [lidar_localization]: Activating end
[lidar_localization_node-1] [INFO] [1747976542.652186905] [lidar_localization]: odomReceived
[lidar_localization_node-1] [WARN] [1747976542.652191823] [lidar_localization]: odom time interval is too large
[lidar_localization_node-1] [INFO] [1747976542.652208097] [lidar_localization]: odomReceived
[lidar_localization_node-1] [INFO] [1747976542.670647528] [lidar_localization]: odomReceived
[lidar_localization_node-1] [INFO] [1747976542.691682847] [lidar_localization]: odomReceived
[lidar_localization_node-1] [INFO] [1747976542.711162512] [lidar_localization]: cloudReceived
[lidar_localization_node-1] lidar_localization_node: ./kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:239: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, unsigned int, pcl::Indices&, std::vector<float>&) const [with PointT = pcl::PointXYZI; Dist = flann::L2_Simple<float>; pcl::Indices = std::vector<int>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
[ERROR] [lidar_localization_node-1]: process has died [pid 53902, exit code -6, cmd '/home/dandelion/localization_ws/install/lidar_localization_ros2/lib/lidar_localization_ros2/lidar_localization_node --ros-args -r __node:=lidar_localization -r __ns:=/ --params-file /home/dandelion/localization_ws/install/lidar_localization_ros2/share/lidar_localization_ros2/param/localization.yaml -r /cloud:=/livox/lidar'].
```

## Additional information

I changed `use_odom` to `false`. There is no problem.

rsasaki0109, 2025-05-26:
Thanks for the report.

To be honest, I'm not sure what the exact cause is at this point. It might be something I could figure out with some investigation, but I don't currently plan to look into it.

Since it works with use_odom:=false, I'd suggest sticking with that for now.

rsasaki0109, 2025-05-26:
That said, I asked GPT about the issue, and it pointed out two things that might be related:

odomReceived() might be called before the initial pose is set, and corrent_pose_with_cov_stamped_ptr_ could be null. Adding a check like this might help:

```cpp
if (!corrent_pose_with_cov_stamped_ptr_) {
  RCLCPP_WARN(this->get_logger(), "Current pose is not initialized yet. Skipping odom update.");
  return;
}
There might be NaN or Inf values in the pose used as the initial guess for align(). A check like this could catch it early:
```

```
if (!std::isfinite(corrent_pose_with_cov_stamped_ptr_->pose.pose.orientation.w)) {
  RCLCPP_WARN(this->get_logger(), "Current pose has invalid (NaN/Inf) values.");
  return;
}
```
I haven't verified these yet, but they might be worth trying if you're looking into it.

atinfinity, 2025-05-27:
I tried your proposed workaround. As a result, this problem is not resolved.

atinfinity, 2025-05-29:
Thank you for your comment. I will investigate.

RACC-MAN, 2025-06-04:
I was also troubled by a similar error on humble version, but after modifying the code as follows, the error no longer occurred.
 
 ## lidar_localization_component.hpp
 ```
  bool use_odom_{false};
  double last_odom_received_time_;
  bool use_imu_{false};
  bool enable_debug_{false};
  
  bool odom_availalable_{false} // add boolean variable

  int ndt_num_threads_;
  int ndt_max_iterations_;

 ```
 
 ## lidar_localization_component.cpp
 ```
  map_sub_ = create_subscription<sensor_msgs::msg::PointCloud2>(
    "map", rclcpp::QoS(rclcpp::KeepLast(1)).transient_local().reliable(),
    std::bind(&PCLLocalization::mapReceived, this, std::placeholders::_1));

  //odom_sub_ = create_subscription<nav_msgs::msg::Odometry>(
  //  "odom", rclcpp::SensorDataQoS(),
  //  std::bind(&PCLLocalization::odomReceived, this, std::placeholders::_1));

  cloud_sub_ = create_subscription<sensor_msgs::msg::PointCloud2>(
    "cloud", rclcpp::SensorDataQoS(),
    std::bind(&PCLLocalization::cloudReceived, this, std::placeholders::_1));
 ```
 delete initialization of odom_sub_ in PCLLocalization::initializePubSub() function

 ```
  if (!map_recieved_ || !initialpose_recieved_) {return;}
  if(!odom_available_)
  {
    odom_available = true;
    odom_sub_ = create_subscription<nav_msgs::msg::Odometry>(
      "odom", rclcpp::SensorDataQoS(),
      std::bind(&PCLLocalization::odomReceived, this, std::placeholders::_1));

  }
  RCLCPP_INFO(get_logger(), "cloudReceived");
 ```
 then, insert initialization of odom_sub_ in PCLLocalization::cloudReceived().
 
 ## Possible Cause
 Because the instantiation of 'odom_sub_' is too earlier than that of 'cloud_sub_'.