rsasaki0109/lidar_localization_ros2 #65: [Bug] Robot pose tilted when using tilted LiDAR.

abudori3939, 2025-07-11:
# Summary
When using a physically tilted LiDAR (e.g. base_link -> livox_frame transformation defined in robot_state_publisher), lidar_localization_ros2 seems to perform calculations without taking this tilt into account. As a result, the final robot pose estimate (map -> base_link) is incorrectly rotated, effectively applying the tilt twice.

# Reproduced environment
TF tree: map -> odom -> base_footprint -> base_link -> livox_frame

[image]

Discription: A static transformation between base_link and livox_frame is defined (e.g. via URDF and robot_state_publisher) to represent the physical tilt of the LiDAR (e.g. tilted 35 degrees forward).

lidar_localization_ros2 subscribes to the PointCloud2 topic /livox/lidar (frame_id: "livox_frame")

[image]

# Observed behavior
After localization, the input point cloud appears overly tilted when visualized in a map frame.

1. Point cloud in map frame (/livox/lidar)

[image]

Discription: The ground, which should be horizontal, is tilted in the opposite direction to the sensor tilt.

2. Raw point cloud in livox_frame

[image]

Discription: This is the correct raw data from the sensor's perspective. The ground appears tilted because the LiDAR is tilted forward.

3. Point cloud transformed to base_link frame

[image]

Discription: This is the point cloud after transformation by TF from robot_state_publisher. The sensor tilt has been correctly corrected and the ground is horizontal.

# Cause analysis
The cloudReceived callback of lidar_localization_ros2 is performing registration with a tilted point cloud. The published pose is the base frame, so registration needs to be performed with a point cloud corrected to the base frame coordinate system.

The tilted point cloud (point cloud from the LiDAR's viewpoint) is scan matched with the 3DMap, and the obtained posture is sent as the base frame. The robot_state_publisher converts from the base frame to the livox_frame, so the tilt correction is applied twice in total.

[image]

# Expected modifications
Before the alignment step, the received point cloud frame is rotated to the base frame. This ensures that localization is always performed based on the level base_link, regardless of the LiDAR mounting direction. This rotation requires obtaining the rotation matrix from TF and calculating it with PCL.

rsasaki0109, 2025-07-11:
Thank you for the detailed report.
You’re right, the package I created only considered horizontally mounted LiDAR.
I think the following reference might be helpful:
https://github.com/autowarefoundation/autoware_core/blob/main/localization/autoware_ndt_scan_matcher/src/ndt_scan_matcher_core.cpp#L376

rsasaki0109, 2025-07-11:
Thank you so much for taking the time to provide such a detailed report. I’m really sorry, but to be transparent, I don’t have much motivation to work on this right now, so I can’t promise when I’ll be able to look into it.

abudori3939, 2025-07-11:
Thank you for reading my report.
After much thought, I came up with the same method as in the document you mentioned (converting from base frame to sensor).
I have already been able to perform localization correctly with the tilted sensor on my local PC.

Thanks to your great software, I have achieved most of what I wanted to do.
I can make a pull request for the corrections I applied to the tilted sensor, so please take a look if you'd like.

commit fb925743 (2025-07-16): [Fixes #65] Transform input point cloud to base_frame to support tilted LiDAR (#66)

* 変更：PCDマップ読み取りの場所で、PLYファイルも読み取れるようにした

* 追加：map->odomのTF変換モードを追加

* 修正：map->odomのTFのタイムスタンプ漏れを解消

* 追加：点群をbase_frame座標系に直し、取り付け位置で歪ませないロジックを追加
closed 2025-07-16 by rsasaki0109